j0ono0/pinout #1: Include some example output diagrams in the README?

mithro, 2021-03-15:
It would be awesome if you included some output diagrams examples in the README.

j0ono0, 2021-03-17:
Including some sample diagrams has prompted an immediate focus on better documentation in general. a "hero" pinout diagram has been included at the start of README to inform/inspire readers of potential outcomes along with graphical explanations supporting/aiding the tutorial documentation that has been moved to readthedocs.

Some project scope expansion ideas have come to mind based on this issue: more visual examples showing possible design variations along with preconfigured style settings (I see the byo css as a barrier to user success).

!! ~~ ** many thanks mithro for the input ** ~~ !!

closed 2021-03-17 by j0ono0

mithro, 2021-03-17:
This look awesome! Thanks for your work @j0ono0

umarcor, 2021-03-18:
@j0ono0 I added this project to https://hdl.github.io/awesome/items/. You might want to provide some brief (or long) description: https://github.com/hdl/awesome/blob/develop/content/items/pinout.md.

Some other hints:

- On the right sidebar of https://github.com/j0ono0/pinout, click on the gear (settings) and you can add the URL to the docs there. You can also add tags/topics.
- In the README, you can use HTML for centering the images. See, for instance, https://github.com/im-tomu/fomu-workshop.
- You can add shields/badges (as in fomu-workshop). I suggest using https://shields.io/.

j0ono0, 2021-03-19:
Thanks for the hints @umarcor. I don't know how long I'm allowed to hold onto a 'newbie' status for github but I'm still feeling like one. 
And *big thanks* for the list inclusion! Very flattering to share list space with some other titles I spied in there. I'll endeavour to add a long description soon.