j0ono0/pinout #7: Title / text block

j0ono0, 2021-03-23:
Enable a way for other supporting text to be added to the diagram.

arturo182, 2021-05-09:
Would love to be able to do `diagram.add_text()` and even `diagram.add_raw()` (name pending 😉) to add raw svg code that will be appended to the diagram :)

j0ono0, 2021-05-09:
> 
> 
> Would love to be able to do `diagram.add_text()` and even `diagram.add_raw()` (name pending 😉) to add raw svg code that will be appended to the diagram :)

Hi arturo182, thanks for the interest in pinout! I've had a pretty amazing few days watching twitter, the support and interest has been super rewarding.

**diagram.add_text()** is high on my list as I would like to add title blocks and notes to diagrams - typical of what might appear on a pinout diagram.  Are their specific features you would like to see a text component support?

**diagram.add_raw()** is a great idea to open 'svg mashups' to advanced users. Currently adding an svg image and embedding it (ie: diagram.add_image("my-svg-file.svg", embed=True) simple inserts the SVG file markup so might be used as a work around. ...I'm getting some ideas on this as I type and might be trivial to do - could make it into the v0.0.9 release.

j0ono0, 2021-05-23:
0.0.10 will include a 'textblock' component. It is unsophisticated but capable of inserting multi-line text content into the diagram. CSS styling access is excellent via use of the <tspan> tag.

PyCairo was investigated in detail as a solution but discarded as cross platform text support (ie dev setup for windows) proved to be discouragingly complex.

closed 2021-05-23 by j0ono0