j0ono0/pinout #10: Build GitHub actions based CI to publish to PyPI on releases

whatnick, 2021-04-03:
I would like to contribute by building a GitHub actions based CI to publish to PyPI on release tags. As a pre-requisite for this to ensure code quality and stability an appropriate PyCQA suite and Testing suite is necessary.

j0ono0, 2021-04-07:
Sounds great. I'll need to revise my general development and QA processes to match with the automation (A good thing).

umarcor, 2021-05-24:
For reference (https://github.com/VUnit/vunit/blob/master/.github/workflows/push.yml#L193-L227):

```yml
  deploy:
    runs-on: ubuntu-latest
    if: github.event_name == 'push' && startsWith(github.ref, 'refs/tags/v')
    steps:

    - name: '🧰 Checkout'
      uses: actions/checkout@v2
      with:
        submodules: recursive

    - name: '🐍 Setup Python'
      uses: actions/setup-python@v2
      with:
        python-version: 3.9

    - name: '🐍 Install dependencies'
      run: |
        pip install -U pip
        pip install -U setuptools wheel twine
        
    - name: '🚀 Build and deploy to PyPI'
      if: github.repository == 'j0ono0/pinout'
      env:
        TWINE_USERNAME: __token__
        TWINE_PASSWORD: ${{ secrets.DEPLOY_TOKEN }}
      run: |
        python setup.py sdist
        twine upload dist/*
```