j0ono0/pinout #11: [Discussion] Decide on coding styles and apply PyCQA + Pytest

whatnick, 2021-04-03:
To improve consistency and readbility with external contributions let us add
- A [contributors file](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/setting-guidelines-for-repository-contributors)
- Add some combination of linting and code-quality analysis via CI
- Start working on a small test suite to check stability on PR's

I would work on all these as time allows.

j0ono0, 2021-04-07:
They all sound like things I should read up on!
I'm regularly rewriting large chunks of the core code - might be a good time to learn, and implement, testing processes and try out some linting.

j0ono0, 2021-04-07:
@whatnick any recommendations on style/linting to conform to?

whatnick, 2021-04-08:
Typical PyCQA settings are
- A pre-commit hook on developer machines to reject failing code
- A lint github action on push / on PR to maintain clean main branch from external contributions
- Shed is a fully opinionated option, most projects I work on use isort, flake8, black and pylint. Pylint is perhaps the heaviest lift.

j0ono0, 2021-04-12:
Checked out pylint - annoying. pedantic. -but- I can see the benefit and think It'll be worth enforcing style consistency with it. I've only read-up on isort but it also sounds like a good move - if only because I occasionally waste time pondering import sort order!

I'll see if I can get my current dev branch (0.0.9) into a passable format.