j0ono0/pinout #14: SVG board image not embedding properly

nobodywasishere, 2021-04-27:
Hello,

Thank you for creating pinout! I just wanted to let you know of an issue I am currently running into an issue when embedding an SVG board image. After setting `embed=True` when running `add_image(...)`, I'm getting an XML error in the output file:
[image]

Commenting out this line gives a scaled down version of the board:
[image]

It should look something like this (this is when embed is False and when using a png board file):
[image]

I do want to embed it though, as it makes it easier for moving it around (and due to CORS). All the code is currently on my github [here](https://github.com/nobodywasishere/upduino-pinout). Let me know if there's anything I can do.

j0ono0, 2021-04-28:
Hi @nobodywasishere, thank you for trying out pinout! Seeing it used is amazingly rewarding.
I've cloned your repo, duplicated the problem and think I've found the issues going on.

**1) XML parsing error:** embedding the SVG image results in the file being directly pasted into the exported file, it simplistically assumes the file is *just* an <SVG> tag and markup with no other file information - as you've discovered the <XML> tag causes the error. Short work around is to comment it out (or remove entirely). I've made a note to address this issue so should not be an problem in future releases.

**2) Sizing issue:** The supplied SVG image still retains its own dimension information in the final exported SVG (it is not overridden by pinout's markup). Consequently the image has to be supplied at a 1:1 ratio for predictable results. Resizing the image in inkscape and resetting the document size did the trick in my tests.

Just occurred to me I can amend and send a pull request! which I'll do today --sorry github as a collaboration tool is a bit new to me

FYI: the new version (0.0.9) is in the works. It is not backward compatible but I think a big leap forward in usability and worth upgrading when it is officially released.

Thanks again for making contact,
John.

j0ono0, 2021-05-09:
v0.0.9 removes the <xml> tag when processes SVG files

closed 2021-05-09 by j0ono0