j0ono0/pinout #16: Random thoughts on 0.0.9

arturo182, 2021-05-09:
These are a few random thoughts I had while working with 0.0.9, not sure how useful they are, so take them as you will, 0.0.9 is great as-is anyways! :)

Feels a bit weird to have YAML and the py files, in the main branch the data is in py and just the style stuff is in CSS, that makes sense to me.
It also feels weird that half of the legend info (i.e. position) is in py but then width and radius are in YAML. I guess my confusion comes from the fact that the YAML mixes two types of information: styles (colors, fonts, etc.) and data (tag names).

It feels a bit weird that the tags info is spread between YAML and py, I get that the `categories` param for `add_legend` is optional and only useful if you want to hide some tags, but if you want to add a new tag, you have to do it in two places. Maybe it would be better to add a `visible` property to the tags in YAML, have them be `true` by default s you only set `false` for the ones you want to hide.

I wish I could provide colors as #112233 hex values for the tags :D

I was surprised to find out that the labels don't adjust their width to fit the whole text they contain, but I do understand it might be hard to calculate that dynamically with custom fonts, weights, and all that. Not a big deal.

It would be great to be able to provide background AND text colors for tags as I did in my fork. Some colors don't work as well with black text, but some do.

Can't wait to be able to add custom text in random places!

+ the stuff I did in my work: parallelogram labels, a bullet at the end of the leaderline, hiding leaderlines between labels, having all the leaderlines be one color, and injecting raw svg code

Thanks!

arturo182, 2021-05-09:
Two more:

1. On the main branch you can use CSS to add a margin to the diagram so there is some breathing room in the SVG, not sure if there's a way to do this in 0.0.9
2. Would love an option to export as png, with specified background color (or transparent), specified margins, and size.

j0ono0, 2021-05-10:
Thanks for giving pinout a workout. The feedback and real user experience is invaluable. 
I'll type out some thoughts here - to help clarify things to myself as much as a response to your thoughts (so please excuse rambling)...

_[edit] rambling finished - I'll do a second post for the small stuff_

**TL;DR: The confused implementation reflects early project indecision. Critical feedback and more iterations are needed and under way.**

YAML: Initial thoughts when I was trying this out was  **ALL** user data (content and styling) could be supplied as yaml - A user could generate a diagram from the command-line, no python script at all. I'm not sold on my own idea and currently I've settled on 'configuration' in yaml and content in python. As a hangover from my initial thought, the division not well defined. The legend content is an example of trying out content-as-yaml, I like the concise and clear format but thematically it is confusing and needs resolving one way or the other. All this might be heading down the wrong track though - possibly everything should described in native python with 'cookbook' suggestions of alternative data handling. The user could then choose their mix to suit.

YAML and inline styles: "path of least resistance" resulted with yaml data being inserted as inline attributes into templates effectively removing the css requirement. Parsing yaml to css and implementing as an embedded or external asset has merits for online destinations and is a serious consideration.

CSS: There is still an important place for CSS in pinout, it's removal entirely has been a 'friendly fire' incident - the 'tags' (used as css classes) will be reinstated at some point. I envisage advance users might further style or add interactive aspects  for browser based destinations.

j0ono0, 2021-05-10:
off the top of my head...
text color in labels can be controlled in config:
```
pinlabel:
    label:
        text:
            fill: [<R>, <G>, <B>]
```

hex colors: yes, agreed it should go in. I semi-looked into this already investigating alpha transparency. abandoned progress on discovering SVG doesn't support alpha in colors (eg rgba and the hex equivalent). I should auto detect the alpha value and populate fill-opacity attribute too.

padding on the edge: yes, needs addressing to make output a 'finished document'. poor-work-a-round would be to place the diagram in a wrapper component. wrapper sits at (0,0) whilst diagram can be moved to (x,y) as suits.
`
wrapper = diagram.Component()
# diagram.Diagram inherits from Component under the hood
diagram = diagram.Diagram(x=10, y=10)
wrapper.children.append(diagram)
`

png export: agreed, it'd be useful for me too. Research so far suggests pycairo is the go-to package for this. Open to suggestion on this.

Widths and text: It doesn't seem possible to determine the width of text accurately so elements do not auto adjust to fit. potential solutions I've thought about:
+ Embedding a font and documenting character dimensions
+ Moving the project into browser based language/environment
+ Rendering and measuring behind the scenes - another reason to include pycairo?
In the mean time it's manual input :(

davidoccam, 2021-05-16:
In order to alleviate the color definition issue in config.yaml I have included the following suggestion
HTML color names
The extended colors are the result of merging specifications from 
HTML 4.01, CSS 2.0, SVG 1.0 and CSS3 User Interfaces (CSS3 UI)

#Pink colors
color: &MediumVioletRed         [199, 21, 133]
color: &DeepPink                [255, 20, 147]
color: &PaleVioletRed           [219, 112, 147]
color: &HotPink                 [255, 105, 180]
color: &LightPink               [255, 182, 193]
color: &Pink                    [255, 192, 203]

#Red colors
color: &DarkRed                 [139, 0, 0]
color: &Red                     [255, 0, 0]
color: &Firebrick               [178, 34, 34]
color: &Crimson                 [220, 20, 60]
color: &IndianRed               [205, 92, 92]
color: &LightCoral              [240, 128, 128]
color: &Salmon                  [250, 128, 114]
color: &DarkSalmon              [233, 150, 122]
color: &LightSalmon             [255, 160, 122]

#Orange colors
color: &OrangeRed               [255, 69, 0]
color: &Tomato                  [255, 99, 71]
color: &DarkOrange              [255, 140, 0]
color: &Coral                   [255, 127, 80]
color: &Orange                  [255, 165, 0]

#Yellow colors
color: &DarkKhaki               [189, 183, 107]
color: &Gold                    [255, 215, 0]
color: &Khaki                   [240, 230, 140]
color: &PeachPuff               [255, 218, 185]
color: &Yellow                  [255, 255, 0]
color: &PaleGoldenrod           [238, 232, 170]
color: &Moccasin                [255, 228, 181]
color: &PapayaWhip              [255, 239, 213]
color: &LightGoldenrodYellow    [250, 250, 210]
color: &LemonChiffon            [255, 250, 205]
color: &LightYellow             [255, 255, 224]

#Brown colors
color: &Maroon                  [128, 0, 0]
color: &Brown                   [165, 42, 42]
color: &SaddleBrown             [139, 69, 19]
color: &Sienna                  [160, 82, 45]
color: &Chocolate               [210, 105, 30]
color: &DarkGoldenrod           [184, 134, 11]
color: &Peru                    [205, 133, 63]
color: &RosyBrown               [188, 143, 143]
color: &Goldenrod               [218, 165, 32]
color: &SandyBrown              [244, 164, 96]
color: &Tan                     [210, 180, 140]
color: &Burlywood               [222, 184, 135]
color: &Wheat                   [245, 222, 179]
color: &NavajoWhite             [255, 222, 173]
color: &Bisque                  [255, 228, 196]
color: &BlanchedAlmond          [255, 235, 205]
color: &Cornsilk                [255, 248, 220]

#Green colors
color: &DarkGreen               [0, 100, 0]
color: &Green                   [0, 128, 0]
color: &DarkOliveGreen          [85, 107, 47]
color: &ForestGreen             [34, 139, 34]
color: &SeaGreen                [46, 139, 87]
color: &Olive                   [28, 128, 0]
[... 89 lines cut ...]
color: &SlateGray               [112, 128, 144]
color: &Gray                    [128, 128, 128]
color: &LightSlateGray          [119, 136, 153]
color: &DarkGray                [169, 169, 169]
color: &Silver                  [192, 192, 192]
color: &LightGray               [211, 211, 211]
color: &Gainsboro               [220, 220, 220] 

Usage
tags:
    # <tag>
    #   title: <title>
    #   color: [<0-255>, <0-255>, <0-255>]

    analog:
      title: Analog
      color: *Silver

    dac:
      title: DAC
      color: *MediumPurple

    comms:
      title: Communication
      color: *SandyBrown .....etc.

j0ono0, 2021-05-17:
Nice extension with colours. The circles I'm in make no use of the pre-named colours available in web so I'm a bit blind to their use or potential popularity to other users. I had no idea there were even so many of them! 

I have some similar-ish styling ideas but they are a long way off with other fundamental changes getting priority - The crux of them involves providing a library of stylistic 'themes' and tools to help with legibility and accessibility.

j0ono0, 2021-08-06:
I think reinstating css and removing yaml has resolved much of confusion. A few revisions later I'm still feeling it was the right move too.
Closing this one but feel free to open new issues to reraise aspects of these (or new) thoughts.

closed 2021-08-06 by j0ono0