j0ono0/pinout #21: quick_start instructions do not work with Python 3.8.6

davidoccam, 2021-05-14:
tried to follow instructions. 
"Open a command line in the location you plan to work and enter the following:
py -m pinout.file_manager --duplicate quick_start"

I do not know how to fix this or what is complaining

$ py -m pinout.file_manager --duplicate quick_start

/usr/bin/py:16: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.9 it will stop working (not working in 3.8.6!)
  from collections import Iterable
usage: py [-x] [-l] [-c PRE_CMD] [-C POST_CMD] [-V] [-h] [expression]
py: error: unrecognized arguments: -m --duplicate quick_start

$ python --version
Python 3.8.6

j0ono0, 2021-05-14:
Hi David,
Thank you for trying pinout and I'm sorry that things are not working as they should!

I have not seen these errors before and will need to do some research. I'll cover some basic troubleshooting ideas for you to tryout in the hope it is something simple...

**Basic trouble-shooting** actions to tryout (sorry if these are overly obvious):
- If you are using a virtual environment ensure it is active
- ensure pinout is installed by importing it from a Python REPL
- exchange 'py' with 'python' in the terminal (ie $ python -m pinout.file_manager --duplicate quick_start)

**A potential work-a-round** if you are sure pinout is installed correctly:
Duplicating the files can be 'side-stepped' by cloning the github repository and navigating to: '_pinout/resources/quick_start_'. If an error is in pinout.file_manager generating a diagram should still be possible.

Please let me know how you go with these suggestion and I'll do some further research as well.

-- John.

davidoccam, 2021-05-14:
probably my bad! not using a virtual environment

python -m pinout.file_manager --duplicate quick_start
quick_start_config.yaml duplicated.
quick_start_hardware.png duplicated.
quick_start_pinout.py duplicated.
thanks for the suggestions, I can now try new release.

j0ono0, 2021-05-15:
Great news, thanks for reporting back. I suspect launching with 'python' instead of 'py' was the fix (I'm guessing you are on a Linux OS). The latter is the "Windows way" to launch python - being the OS I'm developing in. I've been a little lazy in documentation and should added a note into the quick-start section about this.

At any rate, using a virtual environment is highly recommended. Pinout is young and evolving rapidly. Consequently new releases regularly have breaking changes. Using a virtual environment for each new project safe-guards against new releases breaking your existing projects.

closed 2021-05-15 by j0ono0