j0ono0/pinout #23: SVGs of Arduino boards

umarcor, 2021-05-24:
As commented in Twitter (https://twitter.com/dcuartielles/status/1396565190007824385) there are some SVGs of Arduino boards available in Fritzing sources/packages: https://github.com/fritzing/fritzing-parts/tree/develop/svg/core/breadboard. Some other are available in the Arduino documentation site, but not all of the boards provide "Fritzing Files". See, for instance, https://docs.arduino.cc/hardware/nano-every and https://docs.arduino.cc/hardware/nano.

---

I was about to try using one of those with pinout, but the quick start is not working for me:

```sh
# git clone https://github.com/j0ono0/pinout
# cd pinout
# PYTHONPATH=$(pwd) python -m pinout.file_manager --duplicate quick_start
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:/msys64/mingw64/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "T:\pinout/pinout/file_manager.py", line 3, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```

j0ono0, 2021-05-24:
That doesn't look good!
I was under the impression pkg_resources ships with Python (please note, I'm am ***not*** a Python expert!)
Initial investigation suggests (re)installing setuptools `pip install setuptools` might help.
Are you using a virtual environment? Easy to set up and highly recommended to avoid messing with your main Python install.

As a backup plan, clone the git repo and access the sample files in _pinout/resources/_

Please post back with how you go with this -- I might need to review how I'm duplicating files if no easy solution is at hand.
-----------
*spoiler alert* In the github repo (0.0.10 branch) is a new _pinout/samples/_ folder that has an almost finished Arduino styled example. It is using the yet to be released version 0.0.10. Today's goal is to finish and post results to Twitter :D

closed 2021-08-06 by j0ono0

umarcor, 2021-08-09:
@j0ono0 I'm sorry I missed your previous message. I can confirm that the current 'main' Quickstart works on Windows using MSYS2 (MINGW64):

```sh
# pacman -S mingw-w64-x86_64-python-cairo
# git clone https://github.com/j0ono0/pinout
# cd pinout
# mkdir test
# cd test
# PYTHONPATH=$(pwd)/.. python -m pinout.manager --duplicate quick_start
data.py duplicated.
hardware.png duplicated.
pinout_diagram.py duplicated.
styles.css duplicated.
# PYTHONPATH=$(pwd)/.. python -m pinout.manager --export pinout_diagram diagram.svg
'diagram.svg' exported successfully.
```

I am not using any virtual environment. However, the installation might be particular because some Python packages are installed through pacman, instead of pip.

Anyway, as said all seems correct now.

j0ono0, 2021-08-10:
I feel like I knew you had resolved the issue - might have been a post on twitter or else where.
anyway - glad to have confirmation that it worked :)