j0ono0/pinout #38: ideas for image x-y positional documentation

gojimmypi, 2021-08-04:
I had somewhat of a difficult time getting started with my [first board](https://github.com/ulx3s/ulx3s-pinout/). The issue was a repeated `edit file`, `run pinout`, `reload image` process as I tried to position `pinout` items on my image with a trial-and-error process. I wrote a little [script](https://github.com/ulx3s/ulx3s-pinout/blob/main/pinout-update.sh) to automate the first two (then manually refresh the browser):

```
#!/usr/bin/env bash

while  [ 1 -eq 1 ]; do 
  inotifywait -e modify styles.css pinout_ulx3s.py data.py
  echo "Change! $(date)"
  rm pinout_ulx3s.svg
  python3 -m pinout.manager --export pinout_ulx3s pinout_ulx3s.svg
done
```
Perhaps this would be useful to others... but I'm also wondering if better documentation on the positioning parameters would help? What do you think of taking the [quickstart](https://pinout.readthedocs.io/en/latest/pages/install.html#quickstart) data file, and having it notated in an image with all the respective data file <x,y> coordinates as defined in the respective data file?

j0ono0, 2021-08-04:
good feedback. I did in fact have a diagram with all required measurements demonstrated on it but it got omitted (for no good reason) along the way.

I'll revise the docs and add it some better explanation regarding positioning. If you struggled other will too.

j0ono0, 2021-08-06:
I had another thought about this topic - _pinout_ is very unfriendly regarding resized images. Current recommendation is to resize images and use them at a 1:1 ratio. BUT this effectively promotes resizing to screen-viewing-sizes and creates a barrier to diagrams setup for print destinations.

Alternative approach I'm thinking of: 
- Cater for image resizing, 
  - Calculate the scaling ratio behind-the-scenes
  - Record 'critical coords' in the Image class
  - Image class returns _scaled_ 'critical coords' on request

Whilst I leave users to 'count pixels' and find their own coordinates on their image, a tool that outputs a grid overlay on an image might help users guess-timate coords with no other tools required... ?

j0ono0, 2021-08-13:
Feature has been implemented -- 0.0.14 (next release) will cater for scaling of images and adding coordinates to them.
This makes the scripting process a little longer but clearly defines the step of measuring/recording key coordinates with the image. Coordinates are also scaled with image scaling providing a more robust UI.

j0ono0, 2021-08-19:
0.0.14 released!
tutorial also updated with some diagrams and a dedicated 'measuring up' step.
initial testing has proven (in my opinion) the change to be a great addition to usability.

closed 2021-08-19 by j0ono0