j0ono0/pinout #39: Differential pair pins: require unique label?

j0ono0, 2021-08-04:
I'm attempting to find and ideal way to document differential pair pins with _pinout_.

If you have experience with them and opinions on how you would like to see them documented in a pinout diagram feel free to post and comment on the discussion here :)

gojimmypi, 2021-08-04:
Lattice has an [application note: Using Differential I/O](https://www.latticesemi.com/-/media/LatticeSemi/Documents/ApplicationNotes/UZ/UsingDifferentialIOLVDSSubLVDSiniCE40Devices.ashx?document_id=47960) that has just a dotted line around the pairs:

[image]

Perhaps something similar to "group" the pinout labels?

smunaut, 2021-08-04:
Depends a bit if they are always diff pairs or not to begin with. In a FPGA it's often configurable whether they act as a diff pair or not. 

But in general just having a + / - signs on the respective pins and a common prefix for the name is enough, don't really need anything fancy. It gets more complicated when the pins are _not_ next to each other :/  And yeah, that happens ...

cbalint13, 2021-08-04:
@j0ono0 

Few ideas borrowed from various places:

  1. Appending **_P** **_N** to the pin names (some software routines can even account on this rule)
  2. Highlighting the pins using **+**  **-** extra symbols (plus even _P and _N on the labels)
  3. Add a special symbol on traces (or right on pins) like Altium(TM) CAD does it.

IMO that Lattice(TM) diagram with dotted/dashed lines is counter-intuitive (my mind thinks to some parasitic shielding).

[image]

j0ono0, 2021-08-05:
Thanks for the fast responses and comments everyone! 
some notes of my own (largely in response and extending other comments)...

Label text as an indicator for differential pair pins - Label content is 100% user supplied so this can be catered for as users preprocess content. Python is also well suited to list manipulation via list-comprehensions -- so my feeling is to leave this up to users rather than build in some enforced standard.

I've given some thought to a method of graphically linking pin locations: 
[image]

...technically it would only require a new leader-line but might through up some complexities around how user would document x2 pins for labels.

On further reflection I wondered if compactness is not such a good goal  - especially for complex hardware that is likely to have printed/PDF docs in A4 (or Letter) format. On that thought I mocked up what a full pinout graphic might be:
[image]

By pulling out each set of pins it:
- leaves less potential ambiguity 
- allows ability to include notes for sets of pins
- caters for pins to be labelled independently (ie not as pairs) without re-configuring a preexisting template
- An assumption > would allow pin functions of a programmed FPGA board to be documented, without re-configuring a preexisting template

On capability to achieve such a layout with _pinout_,  I think it might already technically possible, but need try/review/amend before claiming that victory.

So, after a day of pondering it, I feel the latter example might be a better goal to aim for? **Open to other opinions and input to my thinking.**

suarezvictor, 2021-08-05:
A plus por minus sign after the name is simple and clear enough IMO

cbalint13, 2021-08-05:
@j0ono0 

I personally like ```idea A``` (elegant and compact label) with some amends:
  * add +/- extra symbol or _P _N to label
  * longer arrow line to touch booth physical pins/holes (now touches one)
  * add **|** symbol in between physical pins/holes (just like it is between the labels)

Another rule: In 99% of cases diff pair pins will be in the **same places** (as pair) otherwise it fails the impedance constrains on the very PCB, so YES this compactization (idea A) of label pairs can be a main rule. Probably some slow diff signals like RS485 may afford pins that not at the same place but those are labeled as **A** and **B** by whole industry (not even highlighting their differential nature).

cbalint13, 2021-08-05:
> @j0ono0
> 
> I personally like `idea A` (elegant and compact label) with some amends:
> 
> * add +/- extra symbol or _P _N to label
> * longer arrow line to touch booth physical pins/holes (now touches one)
> * add **|** symbol in between physical pins/holes (just like it is between the labels)

Another opinion (last):  Variant C (of arrow) in your second graph/draw solves all my last two amends in most elegant way.

pkatarzynski, 2021-08-09:
I also like the idea of _P and _N naming.

j0ono0, 2021-08-16:
Feedback seems to suggest naming conventions are adequate to communicate the diferential-pair relation. I still haven't ruled out creating some graphical icon down the track. Some twitter posts combined with a recent layout experiment matched up nicely so I thought it worth recording here --

'double-layer' sample generated with pinout (thanks to @somhi for suggesting it):
[image]

Mockup of possible 'joining' icon:
[image]