j0ono0/pinout #44: PNG / SVG export differences

somhi, 2021-08-18:
When exporting there is following differences of text not center aligned inside labels in the SVG version whereas PNG version left is Ok.  Any clues?
[image]

j0ono0, 2021-08-18:
Can you provide a little more info on how you are creating the PNG?

I've see this when testing with Adobe Illustrator and believe it occurs due to a CSS style (_dominant-baseline: middle_) not being supported.

Converting to PNG via Inkscape should work and, as of version 0.0.13, pinout can export to png (just change the file name suffix when exporting). 

**warning**: pinout uses [cairosvg ](https://cairosvg.org/documentation/)to do the conversions which has its own dependencies including GTK3. As a windows user I found instructions on installing GTK3 convoluted! I can provide more details if you are in that same boat.

somhi, 2021-08-19:
I just used the same script for both:
#  python3 -m pinout.manager --export pinout_deca pinout_deca.svg
    python3 -m pinout.manager --export pinout_deca pinout_deca.png
Another user I talked has the same problem as me.
I'm using ubuntu 20.04

j0ono0, 2021-08-20:
Does the error appear on the quick_start files too? (trying to rule out a styling error particular to your project)
to access quick_start files:
```Python
# duplicate the files
python -m pinout.manager -d quick_start

# export the png
python -m pinout.manager -e quick_start_diagram.png
```
I cloned https://github.com/DECAfpga/DECA_board and exported a PNG via Ubuntu on Windows (WSL) but all looked correct. Ubuntu version I have is 16.04.4 **but** I did have to install _libcairo2-dev_ ...makes me wonder if cairo needs updating? info on installing here: https://cairographics.org/download/

I've just found a release note for cairosvg - version 2.5.0 (released October 29, 2020) added support for _dominant-baseline_. Can you check your cairosvg version too? Installed python packages can be listed from the command line: >>> pip freeze

sorry, I feel like I'm piling up chores rather than providing solutions. I'll investigates some alternative method(s) of aligning the text too.

infphyny, 2021-08-21:
I am on Ubuntu 21.04 . The behavior is the same before and after installing cairosvg 2.5.0-1.1.

pip freeze output show CairoSVG==2.5.2

I have tested on quickstart pinout files, text inside label for png work as expected. Text inside label for svg are higher like the image somhi showed .  pinout version used for the test is 0.0.14.

j0ono0, 2021-08-21:
Hi infphyny - thank you for posting results of checking and testing. 

**I've just realized that I've misread which output you are both seeing display errors in !!!** I previously had problems with cairo so had that fixed in my mind.

Display errors in the PNG suggest an issue relating to cairo. Display errors in the **_SVG_** will be related to the application you are viewing them in. My testing (in windows 10):
Passes:
- Firefox 
- Chrome
- Edge 
- InkScape
 
Fails:
- IE 11 
- Adobe Illustrator CS4 (A very old version)
- Gimp

What app(s) are you viewing the SVG with?

infphyny, 2021-08-21:
j0ono0, You are right, the problem is the gnome image viewer which is the default app to open svg on Ubuntu. Opened the file with Firefox and Inkscape and it work on Ubuntu.

So pinout work perfectly, it's an image viewer application issue. Maybe mention that in readme.md that some application have problem viewing svg file.

Next time I will test with many different applications,  lesson learned.

Thanks.

j0ono0, 2021-08-23:
Thanks for following up and posting back. I've added a warning into the installation section of the docs - with the issue cropping up for 2 people there will be others who will have had the same experience.

I'll close the issue - but - if you think you have a use case where this limitation will be a problem let me know and I'll build in a manual way to align text inside pin-labels rather than relying on the 'dominant-baseline' css style.

closed 2021-08-23 by j0ono0